zigpy/zigpy-znp #192: zigpy-znp application examples

sergorl, 2022-10-31:
Hi, guys. Are there some examples like?:

1. Device enumeration: list all connected devices to my zigbee-adapter
2. Data reading from devices
3. Device command sending

jaroslavpsenicka, 2023-01-02:
+1 sergorl 
I'm trying to use the zigpy-znp, I managed to bootstrap the app as it seems:
```
async def run(radio_path):
    config = ControllerApplication.SCHEMA({"device": {"path": radio_path}})
    app = ControllerApplication(config)

    try:
        await app.startup()
    except NetworkNotFormed as e:
        LOGGER.error("Could not start application: %s", e)
        return

async def main(argv):
    parser = setup_parser("TEST")
    args = parser.parse_args(argv)
    await run(args.serial)

if __name__ == "__main__":
    asyncio.run(main(sys.argv[1:]))  
```
and it seems to start fine, then it stops
```
pi@nezha:~$ python3 test2.py /dev/ttyACM0 
2023-01-02 12:49:12.605 nezha zigpy.device INFO [0x0000] Requesting 'Node Descriptor'
2023-01-02 12:49:12.681 nezha zigpy.device INFO [0x0000] Got Node Descriptor: NodeDescriptor(logical_type=<LogicalType.Coordinator: 0>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice|AlternatePanCoordinator: 143>, manufacturer_code=0, maximum_buffer_size=80, maximum_incoming_transfer_size=160, server_mask=1, maximum_outgoing_transfer_size=160, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=True, *is_coordinator=True, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=False, *is_security_capable=False)
2023-01-02 12:49:12.688 nezha zigpy.device INFO [0x0000] Discovering endpoints
2023-01-02 12:49:12.738 nezha zigpy.device INFO [0x0000] Discovered endpoints: [2, 1]
2023-01-02 12:49:12.741 nezha zigpy.device INFO [0x0000] Initializing endpoints [<Endpoint id=2 in=[] out=[] status=<Status.NEW: 0>>, <Endpoint id=1 in=[] out=[] status=<Status.NEW: 0>>]
2023-01-02 12:49:12.747 nezha zigpy.endpoint INFO [0x0000:2] Discovering endpoint information
2023-01-02 12:49:12.819 nezha zigpy.endpoint INFO [0x0000:2] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=2, profile=49246, device_type=2080, device_version=0, input_clusters=[0], output_clusters=[])
2023-01-02 12:49:12.824 nezha zigpy.endpoint INFO [0x0000:1] Discovering endpoint information
2023-01-02 12:49:12.896 nezha zigpy.endpoint INFO [0x0000:1] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=1024, device_version=0, input_clusters=[0, 6, 10, 25, 1281], output_clusters=[1, 32, 1280, 1282])
2023-01-02 12:49:12.903 nezha zigpy.device INFO [0x0000] Already have model and manufacturer info
2023-01-02 12:49:12.906 nezha zigpy.device INFO [0x0000] Discovered basic device information for <ZNPCoordinator model='CC2531, Z-Stack Home 1.2 (build 20211115)' manuf='Texas Instruments' nwk=0x0000 ieee=00:12:4b:00:1c:dc:ab:ab is_initialized=True>
2023-01-02 12:49:12.912 nezha zigpy_znp.zigbee.application WARNING Your network is using the insecure Zigbee2MQTT network key!
2023-01-02 12:49:12.915 nezha zigpy_znp.zigbee.application INFO Permitting joins for 0 seconds
2023-01-02 12:49:12.940 nezha zigpy_znp.zigbee.application INFO Coordinator is not permitting joins anymore
pi@nezha:~$
```
I need to run an endless loop now I reckon?

projetoarduino, 2023-06-02:
Hi @jaroslavpsenicka  you on more complete example, you have ?

Hedda, 2023-10-24:
FYI, this + similar questions about application examples and API documentation have been asked in many other issues, see:

https://github.com/zigpy/zigpy-znp/issues/77

You might also be interested in the somewhat related discussions here:

https://github.com/zigpy/zigpy/discussions/709

https://github.com/zigpy/zigpy/issues/557

https://github.com/zigpy/zigpy/issues/842

https://github.com/zigpy/zigpy/pull/848

https://github.com/zigpy/zigpy-cli/pull/2

As well as be aware of `zigpy-cli` as many zigpy-znp tools are deprecated as have instead been moved to zigpy-cli instead:

https://github.com/zigpy/zigpy-cli

https://github.com/zigpy/zigpy-cli/pull/2