zigpy/zigpy-znp #198: Python3.7

pipiche38, 2023-01-04:
In the https://github.com/zigpy/zigpy-znp/pull/195 you have updated the python_requires from 3.7 to 3.8.
I was under the assumption that the fact that 3.7 was not supported and especially the tests were not maintained any more, but it was not enforced to have 3.8 to get zigpy installed.

We have a number of users which still run on python3.7 for different reasons, and that really a shame to prevent them to use our plugin.

puddly, 2023-01-04:
Run pip with `--ignore-requires-python` and you can install it.

I'm not going to be wasting any more time on Python 3.7 support, considering it is going to be EoL in five months and requires duplicating effort to get unit tests running.

closed 2023-01-04 by puddly

pipiche38, 2023-01-04:
@puddly no problem at all, and thanks for the trick, that is more than enough.