zigpy/zigpy-znp #199: OTA for Texas Instruments CC1352/CC2652, Z-Stack

gielk, 2023-01-30:
I was following the next instructions : [Manually initiating OTA firmware update](https://github.com/zigpy/zigpy/wiki/OTA-Device-Firmware-Updates#new-gui-method)

But the instructions are not defining how and what I need to fill in the required 3 fields.
Can someone tell me how I can find this information?

[image]

```
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.1.7",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.7",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/Amsterdam",
    "os_name": "Linux",
    "os_version": "5.15.90",
    "supervisor": "2023.01.1",
    "host_os": "Home Assistant OS 9.5",
    "docker_version": "20.10.22",
    "chassis": "vm",
    "run_as_root": true
  },
  "custom_components": {
    "spotcast": {
      "version": "v3.6.30",
      "requirements": []
    },
    "dwains_dashboard": {
      "version": "3.3.0",
      "requirements": []
    },
    "adaptive_lighting": {
      "version": "1.4.1",
      "requirements": []
    },
    "tplink_deco": {
      "version": "3.1.0",
      "requirements": [
        "pycryptodome>=3.12.0"
      ]
    },
    "meural": {
      "version": "1.0.1",
      "requirements": []
    },
    "hacs": {
      "version": "1.30.1",
      "requirements": [
        "aiogithubapi>=22.10.1"
      ]
    },
    "localtuya": {
      "version": "5.0.0",
      "requirements": []
    },
    "afvalwijzer": {
      "version": "2023.01.01",
      "requirements": []
    },
    "webrtc": {
      "version": "v3.0.1",
      "requirements": []
    },
    "eufy_security": {
      "version": "6.2.0",
[... 400 lines cut ...]
            "endpoint_attribute": "power",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x0020": {
            "endpoint_attribute": "poll_control",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x0500": {
            "endpoint_attribute": "ias_zone",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x0502": {
            "endpoint_attribute": "ias_wd",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

puddly, 2023-01-30:
You can put `0` for all of them.

gielk, 2023-01-31:
I did this and this it what i saw in the log.
[image]

MattWestb, 2023-01-31:
Mark **only** `QueryJitter` and move the `query_jitter` slider so it getting one no zero value and isue zigbee command shall working OK.

gielk, 2023-01-31:
@MattWestb I did move the query_jitter 
[image]
Still getting an error, tried multiple values.
[image]

MattWestb, 2023-01-31:
What kind of device is you trying doing OTA updating on ?
Is on Signal repeater, router or end device ?
Its looks it being the coordinator the the error is saying its address `0x0000` and its cant being done with OTA it must being flashed from the bootloader or with debug adapter.

gielk, 2023-01-31:
I'm trying to update the Texas Instruments CC1352/CC2652, Z-Stack 3.30+
It is currently on build 20210708, and I have some problems. Some devices keep disconnecting, and I saw that the first thing I should do is update the firmware on the Zigbee Coordinator. So that is what I'm trying to do here.

MattWestb, 2023-01-31:
Then you must flashing the coordinator with one external tool.
Read more here for TI coordinators : https://www.zigbee2mqtt.io/guide/adapters/#notes

closed 2023-02-22 by puddly