zigpy/zigpy-znp #204: Unable to start zigpy-znp

pipiche38, 2023-02-14:
With the recent changes where there is no more package information on the __init__.py file we are not able to start the zigpy_znp layer 

```
Traceback (most recent call last):
  File "/var/lib/domoticz/plugins/Domoticz-Zigbee/Classes/ZigpyTransport/AppZnp.py", line 61, in startup
    await self.initialize(auto_form=True, force_form=force_form)
  File "/var/lib/domoticz/plugins/Domoticz-Zigbee/Classes/ZigpyTransport/AppZnp.py", line 36, in initialize
    await Classes.ZigpyTransport.AppGeneric.initialize(self, auto_form=auto_form, force_form=force_form)
  File "/var/lib/domoticz/plugins/Domoticz-Zigbee/Classes/ZigpyTransport/AppGeneric.py", line 57, in initialize
    await self.load_network_info(load_devices=False)
  File "/var/lib/domoticz/plugins/Domoticz-Zigbee/zigpy_znp/zigbee/application.py", line 160, in load_network_info
    await self._znp.load_network_info(load_devices=load_devices)
  File "/var/lib/domoticz/plugins/Domoticz-Zigbee/zigpy_znp/api.py", line 229, in load_network_info
    await self._load_network_info(load_devices=load_devices)
  File "/var/lib/domoticz/plugins/Domoticz-Zigbee/zigpy_znp/api.py", line 139, in _load_network_info
    package_version = pkg_resources.get_distribution("zigpy_znp").version
  File "/usr/lib/python3.10/site-packages/pkg_resources/__init__.py", line 471, in get_distribution
    dist = get_provider(dist)
  File "/usr/lib/python3.10/site-packages/pkg_resources/__init__.py", line 347, in get_provider
    return working_set.find(moduleOrReq) or require(str(moduleOrReq))[0]
  File "/usr/lib/python3.10/site-packages/pkg_resources/__init__.py", line 891, in require
    needed = self.resolve(parse_requirements(requirements))
  File "/usr/lib/python3.10/site-packages/pkg_resources/__init__.py", line 777, in resolve
    raise DistributionNotFound(req, requirers)
pkg_resources.DistributionNotFound: The 'zigpy_znp' distribution was not found and is required by the application
```

puddly, 2023-02-14:
Is the `zigpy-znp` package installed? Or are you directly importing it as a module?

pipiche38, 2023-02-15:
zigpy-znp is just a result of a git clone in my environment so I can work on it.
The response is them I'm importing the module directly

pipiche38, 2023-02-17:
Do I miss something ?

Taking in consideration that the python code is launched through an embedded python framework in a C++ code, we cannot really use `virtualenv` as the C++ executable is just launching a python interpreter.

puddly, 2023-02-17:
That shouldn't preclude you from being able to install packages, since you install all the other dependencies? Just `pip install -e /path/to/root` it.

pipiche38, 2023-02-20:
Ok.

closed 2023-02-20 by pipiche38