zigpy/zigpy-znp #206: Dubious licensing

gspr, 2023-03-14:
The project's licensing terms are unclear. There's a COPYING file that indicates GPL-3 (or later), but at the same time the docs directory contains several files that are clearly *not* under the GPL (these are copyrighted by Texas Instruments, and are not licensed under the GPL).

Could you please be explicit about which parts of the repository are under the GPL (or any other free/open license), and which are not? Otherwise it becomes highly risky to use this code.

puddly, 2023-03-14:
My understanding is that the GPLv3 explicitly only encompasses "source code" and "software". The documentation is in no way required to build, run, or modify the software (from its source code or from a binary `.whl`), so it seems like it would be excluded by its very nature.

But to be more explicit: the license covers everything but `docs/`.

closed 2023-03-14 by puddly

gspr, 2023-03-14:
> My understanding is that the GPLv3 explicitly only encompasses "source code" and "software". The documentation is in no way required to build, run, or modify the software (from its source code or from a binary `.whl`), so it seems like it would be excluded by its very nature.

Just to be clear: I was in no way implying that those PDFs not being GPL licensed is a problem for the software being licensed under the GPL. I filed the issue because it seems obvious that they're not meant to be GPL licensed, which kinda leaves their licensing unclear. When the licensing of *some* files is unclear, it also leaves the licensing terms of *every* file in the repository in doubt (since one falls back to relying on what one considers implied, "obvious" or guesswork).

In addition, you might want to consider whether the copyright holder for those PDFs allows you to include them in your repository and redistribute them at all (I genuinely have no idea). One suggestion to stay on the safe side would simply be to link to TI's own distribution channel for these documents.

> But to be more explicit: the license covers everything but `docs/`.

Thanks for the clarity. It would be even more helpful if e.g. the README file could clearly state something along the lines of "every file in this repository, except for those in `docs/` is licensed under blabla".