zigpy/zigpy-znp #221: timeout error when read CC2652 Sonoff dongle

kyc1338, 2023-07-28:
I tried to save the firmware of the Sonoff zigbee dongle P and used this command "python -m zigpy_znp.tools.nvram_read COM8 -o nvram_backup.json" in two different PC and a MacBook but have the same error below  :

Traceback (most recent call last):
  File "C:\Users\xxxxx\AppData\Local\Programs\Python\Python311\Lib\site-packages\zigpy_znp\api.py", line 673, in _skip_bootloader
    result = await responses.get()
             ^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\xxxxx\AppData\Local\Programs\Python\Python311\Lib\asyncio\queues.py", line 158, in get
    await getter
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "C:\Users\xxxxx\AppData\Local\Programs\Python\Python311\Lib\site-packages\zigpy_znp\tools\nvram_read.py", line 99, in <module>
    asyncio.run(main(sys.argv[1:]))  # pragma: no cover
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\xxxxx\AppData\Local\Programs\Python\Python311\Lib\asyncio\runners.py", line 190, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "C:\Users\xxxxx\AppData\Local\Programs\Python\Python311\Lib\asyncio\runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\xxxxx\AppData\Local\Programs\Python\Python311\Lib\asyncio\base_events.py", line 653, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "C:\Users\xxxxx\AppData\Local\Programs\Python\Python311\Lib\site-packages\zigpy_znp\tools\nvram_read.py", line 90, in main
    await znp.connect()
  File "C:\Users\xxxxx\AppData\Local\Programs\Python\Python311\Lib\site-packages\zigpy_znp\api.py", line 715, in connect
    self.capabilities = (await self._skip_bootloader()).Capabilities
                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\xxxxx\AppData\Local\Programs\Python\Python311\Lib\site-packages\zigpy_znp\api.py", line 672, in _skip_bootloader
    async with async_timeout.timeout(CONNECT_PROBE_TIMEOUT):
  File "C:\Users\xxxxx\AppData\Local\Programs\Python\Python311\Lib\site-packages\async_timeout\__init__.py", line 129, in __aexit__
    self._do_exit(exc_type)
  File "C:\Users\xxxxx\AppData\Local\Programs\Python\Python311\Lib\site-packages\async_timeout\__init__.py", line 212, in _do_exit
    raise asyncio.TimeoutError
TimeoutError

Please advise any solution to this issue.

Hedda, 2023-10-24:
I am guessing that this now works with a newer release of zigpy-znp as previous releases was not fully updated for Python 3.11

Regardless, suggest that you try using `zigpy-cli` instead -> https://github.com/zigpy/zigpy-cli

See zigpy-cli README.md file -> https://github.com/zigpy/zigpy-cli/blob/dev/README.md#network-commands

HamzaETTH, 2025-04-25:
I have the same issue, zigpy-cli returns the same error.