zigpy/zigpy-znp #223: Unable to form network - No module named form_network

kallisti5, 2023-08-23:
```
$ python -m zigpy_znp.tools.energy_scan /dev/ttyUSB0 
2023-08-23 09:02:34.211 freeframe __main__ INFO Starting up zigpy-znp
2023-08-23 09:02:36.796 freeframe __main__ ERROR Could not start application: 
2023-08-23 09:02:36.797 freeframe __main__ ERROR Form a network with `python -m zigpy_znp.tools.form_network`
```

```
$ python -m zigpy_znp.tools.form_network /dev/ttyUSB0 
/usr/bin/python: No module named zigpy_znp.tools.form_network
```

:-\

puddly, 2023-08-23:
The zigpy-znp network formation tools are deprecated, use zigpy-cli instead: https://github.com/zigpy/zigpy-cli

tube0013, 2023-08-23:
use https://github.com/zigpy/zigpy-cli

puddly, 2023-08-23:
Als, if you are just trying to perform an energy scan and use ZHA, this information will be in the diagnostic JSON for the integration under an `energy_scan` key:

[image]