zigpy/zigpy-znp #224: zigpy_znp.tools.network_backup fails with asyncio.exceptions.CancelledError on SkyConnect

redlefloh, 2023-09-24:
I am trying to backup my SkyConnect adapter to then migrate to a SONOFF 3.0 stick due to issues I have with the SkyConnect adapter (see https://github.com/home-assistant/skyconnect.home-assistant.io/issues/73#issuecomment-1730808602) and to see if that helps stabalize my HA system. 

Unfortunately the backup with the command:
```
python -m zigpy_znp.tools.network_backup /dev/ttyUSB_SkyConnect -o SkyConnectBackup.json
```
stops with the following error: and the output file remains empty:
```
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/site-packages/zigpy_znp/api.py", line 684, in _skip_bootloader
    result = await responses.get()
             ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/asyncio/queues.py", line 158, in get
    await getter
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/usr/local/lib/python3.11/site-packages/zigpy_znp/tools/network_backup.py", line 122, in <module>
    asyncio.run(main(sys.argv[1:]))  # pragma: no cover
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/asyncio/runners.py", line 190, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/asyncio/runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/asyncio/base_events.py", line 653, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/zigpy_znp/tools/network_backup.py", line 113, in main
    await znp.connect()
  File "/usr/local/lib/python3.11/site-packages/zigpy_znp/api.py", line 726, in connect
    self.capabilities = (await self._skip_bootloader()).Capabilities
                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/zigpy_znp/api.py", line 683, in _skip_bootloader
    async with async_timeout.timeout(CONNECT_PROBE_TIMEOUT):
  File "/usr/local/lib/python3.11/site-packages/async_timeout/__init__.py", line 141, in __aexit__
    self._do_exit(exc_type)
  File "/usr/local/lib/python3.11/site-packages/async_timeout/__init__.py", line 228, in _do_exit
    raise asyncio.TimeoutError
TimeoutError
``` 

I did the following steps:
```docker exec -it homeassistant /bin/bash```
then 
```pip install zigpy-znp```

then I disabled the Zigbee integration in HA
[image]
and then run
```
python -m zigpy_znp.tools.network_backup /dev/ttyUSB_SkyConnect -o SkyConnectBackup.json
```
I am running HA Version 2023.9.2 in a docker container and the /dev/ttyUSB_SkyConnect port is mapped via /dev/serial/by-id/ in compose.yaml as:
```
  homeassistant                                                                                                                                                                   
    container_name: homeassistant                                                                                                                                                                  
    image: "ghcr.io/home-assistant/home-assistant:2023.9.2"                                                                                                                                        
    volumes:                                                                                                                                                                                       
         - [...some volumes...]                                                                                                                          
    devices:                                                                                                                                                                                       
         - "/dev/serial/by-id/usb-Nabu_Casa_SkyConnect_v1.0_[longid]-if00-port0:/dev/ttyUSB_SkyConnect"
```

A full log with -v is [here](https://docs.google.com/document/d/1ooX4X5nMIc6wGKCs9Q2f4i2WnTru_AnYHFQUAa3y2Sk/edit). Any hint or help would be highly appreciated. Thank You!

TheJulianJES, 2023-09-24:
Use the migration feature in HA or [`zigpy-cli`](https://github.com/zigpy/zigpy-cli).
`zigpy-znp` is for TI based adapters. SkyConnect is SiliconLabs/EZSP.

puddly, 2023-09-24:
I would also recommend the ZHA migration button. These tools aren't necessary for normal use.

redlefloh, 2023-09-28:
thanks, and just fyi, I actually haven't yet started the migration since just after I decided to migrate off SkyConnect, for the first time in a long time SkyConnect has been running stable without a crash since I posted this. So I'll give it a bit more time to observe further.

redlefloh, 2023-09-30:
Okay, and only shortly after I wrote the above, the SkyConnect adapter failed again. I now migrated the radio to the Sonoff 3.0 which worked. Side comment, since I am running in a container but not in privileged mode, new USB devices that are plugged into the host computer do not automatically show up in the container and the migration therefore is not quite as straight forward as the migrate button flow suggests. In my case one first needed to map the USB port of the Sonoff adapter in addition to the SkyConnect adapter in the "devices:" section of compose.yaml, then recompose the container and then with both adapters plugged in restart. Then do the migration with the migrate button. Then afterwards remove the USB port mapping for SkyConnect in compose.yaml, then recompose the container again, then unplug the SkyConnect adapter and restart. After that everything worked again with the Sonoff 3.0 adapter. I will close this issue now. Thanks!

closed 2023-09-30 by redlefloh

redlefloh, 2023-10-22:
Just an update in case anyone still follows this issue. It's been a bit over 3 weeks since I migrated from SkyConnect to Sonoff 3.0 since unfortunately SkyConnect had caused problems with the USB port roughly once a day, making zha unreliable and unusable for me.
Without any changes to the rest of my installation, after migrating to Sonoff 3.0, my system has been very stable ever since and I did not have a single zha issue so far. This means the issue is somehow caused by the combination of my system configuration and SkyConnect, but not by my system configuration and ZigBee in general. I'd love to use SkyConnect at some point again, but I am not sure what would need to change where to avoid the issues that I had.