zigpy/zigpy-znp #236: command "zigpy_znp.tools.nvram_read"  Timeout

ohyeah521, 2023-12-13:
python -m zigpy_znp.tools.nvram_read COM3 -o nvram_backup.json

```
Traceback (most recent call last):
  File "C:\Python3\lib\site-packages\zigpy_znp\api.py", line 684, in _skip_bootloader
    result = await responses.get()
  File "C:\Python3\lib\asyncio\queues.py", line 166, in get
    await getter
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Python3\lib\runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Python3\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Python3\lib\site-packages\zigpy_znp\tools\nvram_read.py", line 99, in <module>
    asyncio.run(main(sys.argv[1:]))  # pragma: no cover
  File "C:\Python3\lib\asyncio\runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "C:\Python3\lib\asyncio\base_events.py", line 647, in run_until_complete
    return future.result()
  File "C:\Python3\lib\site-packages\zigpy_znp\tools\nvram_read.py", line 90, in main
    await znp.connect()
  File "C:\Python3\lib\site-packages\zigpy_znp\api.py", line 726, in connect
    self.capabilities = (await self._skip_bootloader()).Capabilities
  File "C:\Python3\lib\site-packages\zigpy_znp\api.py", line 684, in _skip_bootloader
    result = await responses.get()
  File "C:\Python3\lib\site-packages\async_timeout\__init__.py", line 129, in __aexit__
    self._do_exit(exc_type)
  File "C:\Python3\lib\site-packages\async_timeout\__init__.py", line 212, in _do_exit
    raise asyncio.TimeoutError
asyncio.exceptions.TimeoutError
```

stidch, 2024-12-24:
I see the same and since I'm new in this topic, I don't have a clue why this timeout happens. Any hints?

puddly, 2024-12-24:
This is a failure during initial connection. It means your stick isn't responding when zigpy-znp tried to talk to it.