zigpy/zigpy-znp #239: `zigpy_znp.tools.network_restore` fails with `AttributeError`

GsakuL, 2024-01-29:
I have a spare CC2531 I wanted to repurpose for some testing, and for good measure reset key, pan id, ext pan id. Backup works, but restore failed.

```txt
python3 -m zigpy_znp.tools.network_restore -v -i zigbee_network.json  /dev/serial/by-id/usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B00120233FE-if00
```

```log
2024-01-29 09:41:13.477 artemis zigpy_znp.uart DEBUG Connecting to /dev/serial/by-id/usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B00120233FE-if00 at 115200 baud
2024-01-29 09:41:13.477 artemis zigpy.serial DEBUG Opening a serial connection to '/dev/serial/by-id/usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B00120233FE-if00' (115200 baudrate)
2024-01-29 09:41:13.478 artemis zigpy_znp.uart DEBUG Opened /dev/serial/by-id/usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B00120233FE-if00 serial port
2024-01-29 09:41:13.478 artemis zigpy_znp.uart DEBUG Connected to /dev/serial/by-id/usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B00120233FE-if00 at 115200 baud
2024-01-29 09:41:13.479 artemis zigpy_znp.api DEBUG Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=768, SrcAddr=0xFC7D, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=31, SecurityUse=<Bool.false: 0>, TimeStamp=2196227, TSN=0, Data=b'\x18\x56\x0A\x03\x00\x21\xD0\x22\x04\x00\x21\x3D\x0A', MacSrcAddr=0xFC7D, MsgResultRadius=29)
2024-01-29 09:41:13.479 artemis zigpy_znp.api DEBUG Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=768, SrcAddr=0xFC7D, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=28, SecurityUse=<Bool.false: 0>, TimeStamp=2223332, TSN=0, Data=b'\x18\x57\x0A\x07\x00\x21\x72\x01', MacSrcAddr=0xFC7D, MsgResultRadius=29)
2024-01-29 09:41:13.479 artemis zigpy_znp.api DEBUG Received command: ZDO.SrcRtgInd.Callback(DstAddr=0xFC7D, Relays=[])
2024-01-29 09:41:13.479 artemis zigpy_znp.api DEBUG Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=8, SrcAddr=0xFC7D, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=23, SecurityUse=<Bool.false: 0>, TimeStamp=2963558, TSN=0, Data=b'\x18\x5A\x0A\x00\x00\x20\x01', MacSrcAddr=0xFC7D, MsgResultRadius=29)
2024-01-29 09:41:13.480 artemis zigpy_znp.api DEBUG Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=6, SrcAddr=0xFC7D, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=23, SecurityUse=<Bool.false: 0>, TimeStamp=2966553, TSN=0, Data=b'\x18\x5B\x0A\x00\x00\x10\x00', MacSrcAddr=0xFC7D, MsgResultRadius=29)
2024-01-29 09:41:13.480 artemis zigpy_znp.api DEBUG Toggling RTS/DTR pins to skip bootloader or reset chip
2024-01-29 09:41:13.480 artemis zigpy_znp.uart DEBUG Setting serial pin states: DTR=False, RTS=False
2024-01-29 09:41:13.481 artemis zigpy_znp.api DEBUG Radio is alive: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=768, SrcAddr=0xFC7D, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=31, SecurityUse=<Bool.false: 0>, TimeStamp=2196227, TSN=0, Data=b'\x18\x56\x0A\x03\x00\x21\xD0\x22\x04\x00\x21\x3D\x0A', MacSrcAddr=0xFC7D, MsgResultRadius=29)
2024-01-29 09:41:13.481 artemis zigpy_znp.api DEBUG Giving ping task 0.50s to finish
2024-01-29 09:41:13.632 artemis zigpy_znp.uart DEBUG Setting serial pin states: DTR=False, RTS=True
2024-01-29 09:41:13.783 artemis zigpy_znp.uart DEBUG Setting serial pin states: DTR=False, RTS=False
2024-01-29 09:41:13.935 artemis zigpy_znp.api DEBUG Sending request: SYS.Ping.Req()
2024-01-29 09:41:13.941 artemis zigpy_znp.api DEBUG Received command: SYS.Ping.Rsp(Capabilities=<MTCapabilities.APP|UTIL|SAPI|ZDO|AF|SYS: 377>)
2024-01-29 09:41:13.942 artemis zigpy_znp.nvram DEBUG Detecting struct alignment
2024-01-29 09:41:13.942 artemis zigpy_znp.api DEBUG Sending request: UTIL.AssocFindDevice.Req(Index=0)
2024-01-29 09:41:13.947 artemis zigpy_znp.api DEBUG Received command: UTIL.AssocFindDevice.Rsp(Device=b'\x7D\xFC\x00\x00\x04\x08\x00\x02\x02\x01\x15\x00\x00\x00\x78\x00\xBF\x02\x01\xFE\xF3\x05\x7A\xBF\x00\x0D\xF6\xF4')
2024-01-29 09:41:13.949 artemis zigpy_znp.nvram DEBUG Detected struct alignment: False
2024-01-29 09:41:13.949 artemis zigpy_znp.api DEBUG Detected Z-Stack 1.2
2024-01-29 09:41:13.949 artemis zigpy_znp.api DEBUG Connected to /dev/serial/by-id/usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B00120233FE-if00
Traceback (most recent call last):
  File "/usr/lib/python3.10/runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.10/runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "/home/alpollo/.venv/lib/python3.10/site-packages/zigpy_znp/tools/network_restore.py", line 133, in <module>
    asyncio.run(main(sys.argv[1:]))  # pragma: no cover
  File "/usr/lib/python3.10/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.10/asyncio/base_events.py", line 649, in run_until_complete
    return future.result()
  File "/home/alpollo/.venv/lib/python3.10/site-packages/zigpy_znp/tools/network_restore.py", line 125, in main
    await restore_network(
  File "/home/alpollo/.venv/lib/python3.10/site-packages/zigpy_znp/tools/network_restore.py", line 101, in restore_network
    await znp.write_network_info(network_info=network_info, node_info=node_info)
  File "/home/alpollo/.venv/lib/python3.10/site-packages/zigpy_znp/api.py", line 358, in write_network_info
    if not network_info.stack_specific.get("form_quickly", False):
AttributeError: 'NoneType' object has no attribute 'get'
```
the altered zigbee_network.json
```json
{
    "metadata": {
        "version": 1,
        "format": "zigpy/open-coordinator-backup",
        "source": "zigpy-znp@0.12.1",
        "internal": {
            "creation_time": "2024-01-29T09:26:30+01:00",
            "zstack": {
                "version": 1.2
            }
        }
    },
    "coordinator_ieee": "00124b00120233fe",
    "pan_id": "6a8d",
    "extended_pan_id": "f4fa75310bcb2225",
    "nwk_update_id": 0,
    "security_level": 5,
    "channel": 15,
    "channel_mask": [
        15
    ],
    "network_key": {
        "key": "d939f940b462bd2220db0d8a89ce0aba",
        "sequence_number": 0,
        "frame_counter": 15503
    },
    "devices": [
        {
            "ieee_address": "04cd15fffedb3372",
            "nwk_address": "fc7d",
            "is_child": true
        }
    ]
}
```

```txt
Python 3.10.12
zigpy==0.60.6
zigpy-znp==0.12.1
```

puddly, 2024-01-29:
The network backup and restore commands in the zigpy-znp package are deprecated and will be removed. Instead, use zigpy-cli. It has an example in the readme for both.

GsakuL, 2024-02-02:
ah. thank you. it did not occur to me that `zigpy-cli` was the correct one. I thought it was "just a unified tool" for all packages.

I did had problems installing `zigpy-cli` in the venv (python kept saying `zigpy` was a module thus not directly executable), but using a completely clean/new one worked first time.

I did a clean backup using cli, altered IDs and Key, restored it, and it worked as expected.
Thus closing this as user error.

closed 2024-02-02 by GsakuL