zigpy/zigpy-znp #240: CC2674 (CC2674P/CC2674R) and CC1354 (CC1354P) Zigbee Coordinator compatibility with with zigpy-znp?

Hedda, 2024-02-14:
Koenkk mentioned these require new firmware and znp projects from a newer SimpleLink SDK so wonder if compatible with zigpy-znp?

https://github.com/Koenkk/Z-Stack-firmware/discussions/476

Back-story; SMLIGHT has announced/launched CC2674P10 based "SLZB-07p10" USB dongle and "SLZB-06p10" Ethernet/WiFi adapter:

SLZB-07p10 -> https://smlight.tech/product/slzb-07p10/

SLZB-06p10 -> https://smlight.tech/product/slzb-06p10/

Those look to be sold on Aliexpress now as well:

https://www.cnx-software.com/2024/02/13/zigbee-ethernet-wifi-coordinators-usb-adapters-ti-cc2652p7-cc2674p10-wireless-chips/

There is of course also the official "LP-EM-CC1354P10" (CC1354P10 LaunchPad) development kit hardware from Texas Instruments:

https://www.ti.com/product/CC1354P10#hardware-development

PS: CC2674P10 and CC1354P10 chips are readily available so guess will not be long before will also see such adapters from others too:

[CC2674P10](https://www.ti.com/product/CC2674P10) = 2.4GHz radio SoC for Bluetooth 5.2 LE, Thread, Zigbee 3.0 with 48-MHz ARM Cortex-M33, 296KB SRAM, 1024KB Flash.

[CC1354P10](https://www.ti.com/product/CC1354P10) = Multi-Band radio SoC for Bluetooth 5.2 LE, Thread, Zigbee 3.0 with 48-MHz ARM Cortex-M33, 296KB SRAM, 1024KB Flash.

PPS: CC1354P/CC1354P10 is multi-band and supports both 2.4GHz and "Sub-1GHz" (287-351, 359-439, 431-527, 861-1054, 1076-1315).

Hedda, 2024-03-15:
> Back-story; SMLIGHT announced/launched CC2674P10 based "SLZB-07p10" USB dongle and "SLZB-06p10" Ethernet/WiFi adapter:
> 
> SLZB-07p10 -> https://smlight.tech/product/slzb-07p10/
> 
> SLZB-06p10 -> https://smlight.tech/product/slzb-06p10/

Not sure if the webpages on SMLIGHT website has the wrong infoprmation (as long like a direct copy-and-paste from the CC2652P based products) but noticed that they claim on both those webpages that their CC2674P based SLZB-07p10 + SLZB-06p10 ship pre-flashed and are compatible with the ZHA integration in Home Assistant and Zigbee2MQTT out-of-the-box, is that correct?

[image]

[image]

Hedda, 2024-08-05:
Tip: FYI, noticed SMLIGHT is now selling/shipping their CC2674P1 based SLZB-07p10 USB adapter dongle (for just under $15US):

https://smlight.tech/product/slzb-07p10/

Note! They specifically state there on that webpage that "_**ZIGBEE FIRMWARE FOR ZIGBEE CHIP CC2674P10 IS STILL UNDER DEVELOPMENT**_" and that "_**YOU CAN BUY THIS DEVICE ONLY FOR DEVELOPMENT PURPOSES**_".

CC2674P10 GPIOS for developers (corresponds to LAUNCHXL board CC2652P):

RF: RF_24GHZ → DIO28, RF_HIGH_PA → DIO29;
UART: TX → DIO13, RX → DIO12;
LEDs: GREEN LED → DIO6, RED LED → DIO7;
BOOT: BSL → DIO15, Reset → RESET_N;

PS:FYI, JelmerT's upstream [cc2538-bsl](https://github.com/JelmerT/cc2538-bsl/) (programming/flash command-line tool) does not yet support flashing the new CC2674 or CC1354 SoCs but Texas Instruments's SimpleLink SDK looks to include a forked variant of it which someone wrote supposedly should support flashing the newer CC2674 and CC1354 SoCs, see:

https://github.com/TexasInstruments/simplelink-lowpower-f2-sdk/tree/main/source/third_party/cc2538-bsl

Hedda, 2024-08-27:
> PS: FYI, JelmerT's upstream [cc2538-bsl](https://github.com/JelmerT/cc2538-bsl/) (programming/flash command-line tool) does not yet support flashing the new CC2674 or CC1354 SoCs but Texas [...]

FYI, SMLIGHT team has now also released a new web flasher on a cc2538-bsl with added support for CC2674P P10 (CC2674P10):

* https://github.com/smlight-tech/smlight-cc-web-tools

That webflasher depends on their "smlight_cc_flasher" project which in turn is another fork of JelmerT's cc2538-bsl project:

* https://github.com/smlight-tech/smlight-cc-flasher

*  https://pypi.org/project/smlight-cc-flasher/

darkxst (who I believe might have helped written that for them?) wrote a comment that the smlight-cc-flasher code is based on a heavily modified + improved experimental fork of JelmerT's cc2538-bsl project and that tthe fork has been tested with the new [SMLIGHT SLZB-07p10](https://smlight.tech/product/slzb-07p10/) Zigbee Coordinator USB adapter based in CC2674P P10 (CC2674P10) that they released a while ago (but still mark as sold for developers only as there is no well tested community firmware publicly available as of yet):

* https://smlight.tech/flasher/#SLZB-07p10

Also see JelmerT's related issue tracker requesting support for CC2674 (CC2674P/CC2674R) and CC1354 (CC1354P) compatibility:

* https://github.com/JelmerT/cc2538-bsl/issues/160

holsteiner-kiel, 2025-01-11:
Hi. What's up with Support for CC2474? I have an Smlight SLZB06-P10. Think ITS out for months and not supported yet. Is it correct?

Thank you very much!

Hedda, 2026-09-23:
FYI, noticed that Zigbee2MQTT's website now lists many CC2674 (CC2674P10 based) radio adapters as supported under TI zStack adapters:

* https://www.zigbee2mqtt.io/guide/adapters/zstack.html

Suggest ZHA devs test to verify and confirm then update zigpy-znp's README.md to state that Zigbee CC2674 (CC2674P10) is compatible:

* https://github.com/zigpy/zigpy-znp/blob/dev/README.md#hardware-requirements

Suggesting to example try these radio adapters based on Texas Instruments CC2674P10 SoC:

* Sonoff Dongle-PP10 -> https://sonoff.tech/en-eu/products/sonoff-dongle-plus-cc2674p10-zigbee-thread-usb-dongle-dongle-pp10

and

* SMLight SLZB-07p10 -> https://smlight.tech/product/slzb-07p10

These have now been on the marked a while too:

* SMLight  SLZB-06xU (multi-radio network adapter) -> https://smlight.tech/products/slzb-06p10u
* SMLIGHT SLZB-MR3 (multi-radio network adapter) -> https://smlight.tech/product/slzb-mr3/
* SMLIGHT  SLZB-Ultima (multi-radio network adapter) -> https://smlight.tech/global/slzb-ultima

Then there is of course also the "LP-EM-CC1354P10" (CC1354P10 LaunchPad) development kits hardware from Texas Instruments which serve as the official SimpleLink development board for the CC1354P10 and the CC2674P10:

* https://www.ti.com/product/CC1354P10#hardware-development
  * https://www.ti.com/tool/LP-EM-CC1354P10
    * https://www.ti.com/tool/LP-XDS110
  * https://www.ti.com/tool/LP-XDS110ET

Hedda, 2026-11-11:
FYI, developer @CubeZ2mDeveloper who represent Sonoff have submitted this pull request to Home Assistant's core which will enable automatic UCS discovery configuration for SONOFF Dongle-MP10 for the ZHA integration:

* https://github.com/home-assistant/core/pull/188141

His claims here is that they "***have already conducted comprehensive testing of this Dongle-MP10 adapter based on the TI CC2674P10 using the ZHA integration in HA, and only one issue exists: the model is incorrectly identified as CC2652***".

_Originally posted by @CubeZ2mDeveloper in https://github.com/home-assistant/core/issues/188141#issuecomment-6788852071_

"***We are investigating firmware implementation approaches and will submit pull requests to zigpy-znp to correctly identify the model.***"

[image]

[image]

For reference, both Home Assiatant ZHA integration docs and zigpy-znp library zha depends on does not specifically mention [CC2674](https://www.ti.com/product/CC2674P10). See:

* https://www.home-assistant.io/integrations/zha#supported-but-not-recommended-zigbee-adapters

and

* https://github.com/zigpy/zigpy-znp/blob/dev/README.md#hardware-requirements
* 
That is, all docs currently only specifically list CC2652P/CC2652R/CC2652RB-based adapters but not the newer [CC2674 series](https://www.ti.com/product/CC2674P10#order-quality) like CC2674P10.

Perhaps @CubeZ2mDeveloper can submit PRs with updates to both ZHA integration docs and readme for the zigpy-znp radio library?

PS: Again, SMLight who had their CC2674P10-based products on the market a while also they say that those are compatible with ZHA. Ex:

* https://smlight.tech/products/slzb-07p10